andykarpov/karabas-pro #43: Добавить моно режим AY

nihirash, 2021-03-03:
Помимо ABC/ACB ввести еще режим Mono ради страдальцев вроде меня :-)

andykarpov, 2021-03-03:
Хех, беда в том, что я уже пробовал его добавить :)))

Переделок возникло - просто масса:

1) 64-байтную посылку между AVR и FPGA нужно добавить еще один байт для передачи нового состояния (хотя там у @solegstar тоже была хотелка для ремапа буквы дисковода по хоткею - так что придется заводить :)

2) чтобы смикшировать программно 2 16-битных сигнала (сложить их, фактически) нужно делать запас на 1 бит от переполнения, то есть результирующий вектор получится 17-битным. и что потом делать - пока не придумал, наверно отрезать младший бит :)

3) При добавлении новой комбинаторной логики в FPGA проект перестает собираться в работоспособном состоянии. Тут, конечно, связано в роутинге внутри FPGA и в отсутствии constraint'ов по клокам, распространению сигналов и тп :) Я даже отдельную ветку завел с игрищами в этом направлении.

В общем попробую в отдельной ветке, вдруг получится :)

andykarpov, 2021-03-08:
Моно режим добавлен в ветке timequest_constraints. 
Переключается по Menu+F7 (по кругу - ABC, ACB, MONO).

andykarpov, 2021-03-08:
TO BE TESTED :)