andykarpov/karabas-pro #55: Не работает PS/2 клавиатура Perixx PERIBOARD-505

zapolnov, 2021-06-21:
Отладка по UART показала, что клавиатура при подключении посылает постоянно сканкод 0xAA, других сканкодов не посылает.

[image]

С Эволюшеном и ZXDOS+ клавиатура работает.

По результатам обсуждения в чате Карабаса есть мнение, что клавиатура посылает 0xAA с неправильной четностью, ожидая от компьютера ответ 0xFE, чтобы убедиться, что компьютер включен (https://forum.arduino.cc/t/ps2keyboard-library-teensy-and-usb-keyboard-with-ps2-adapter/57549/7).

andykarpov, 2021-06-21:
В проекте используется вот такая простенькая либа для PS/2 клавы: https://github.com/techpaul/PS2KeyRaw
Контроль четности в ней есть, и по-идее на неправильную четность должно ответить как раз 0xFE.
Что еще может не нравиться клавиатуре - сложно сказать.

Попробую переписать код на базе PS2KeyAdvanced от того же автора, код занимает больше места, но и руки больше развязны будут :) С клавой можно будет общаться, посылать состояния светодиодов и другие команды.
Возможно, в ней этого бага нет. Нужно будет поиграться :)

andykarpov, 2021-07-15:
В общем, в ветке avr-ps2keyadvanced закоммичена новая реализация: https://github.com/andykarpov/karabas-pro/tree/avr-ps2keyadvanced/firmware/releases/profi/avr , просьба затестить.
С вероятностью 99% все заработает :)

Для обновления прошивки AVR потребуется программатор, usbasp или аналогичный, с переходником под 6-пиновый разъем. Шить как обычно: `avrdude -c usbasp -p m328p -U flash:w:karabas_pro_revD.hex -U lfuse:w:0xFF:m -U hfuse:w:0xDE:m` (где суффик имени файла прошивки - в зависимости от ревизии платы: revA - для плат ревизии А, revD - для плат ревизий D или DS, без суффикса - для плат ревизии B, C.

closed 2021-08-09 by andykarpov