andykarpov/karabas-pro #90: Реализовать поддержку RTC по стандарту Gluk

andykarpov, 2023-07-19:
Подумать, что делать с пересекающимися портами и реализовать.

Caasper911, 2023-07-19:
Глюк по 'стандарту' включается на доступ одним из битов порта 1FFD, если я не ошибаюсь

andykarpov, 2025-02-10:
Статья: "[CMOS для всех](http://zxdn.narod.ru/hardware/op31cmos.htm)" из электронной газеты Optron #31.

Использование портов, по которым можно было обращаться к часам, стало неким стандартом для ZX-Spectrum и было поддержано в многих программах:
#DFF7 - задаёт номер регистра, с которым мы хотим работать;
#BFF7 - порт данных, через него передаются данные в микросхему/из неё;
#EFF7 - бит 7 этого порта управляет доступом к микросхеме (0 - доступа нет, 1 - доступ есть).