andykarpov/karabas-pro #112: PLEASE HELP KARABAS PRO REV E FOT FLOPPY

nicolaebizi, 2025-07-31:
Greetings
I finished Krabas pro REV E and I was waiting for an integrated for floppy, because I put the net with 2 pieces of rescissed that killed the supply voltage. In the meantime I unsoldered 74ls06 and 74ls38, confirming that 74ls06 has a defective gate. I replaced the integrated and to be sure that no other damage occurred I also replaced EPM3128ATC100. When wd1793 arrived I installed it and found that it did not work.
In the meantime I also made EVGENIY UGRUMOV's emulator, confirming the following:
With a physical unit I managed to write a trd image to the floppy disk.
I can't write anything with Gotec, I think R 64-69 of 330 ohm is influencing something (I soldered them)
The floppy disk was created without error and in the menu =MENU+F2= READS IT BUT DOESN'T EXECUTE IT ....
CAN SOMEONE HELP ME?

closed 2025-08-13 by andykarpov

andykarpov, 2025-08-13:
Also please check the U7 (a level shifter), it contains a signal FDC_STEP. Please check if it is working and converting from LFDC_STEP (3.3v) to FDC_STEP(5v) levels. If not - try to replace that IC.
Some TXB0104 from aliexpress has a broken channels, so maybe that could be a reason the floppy is stuck in the track 0.

Caasper911, 2025-08-13:
Salut Nikolae, dacă ești în Oradea te pot ajuta să testezi Karabas Pro.